xreef/SimpleFTPServer #1: Programmatically find connect status.

thebluecrab, 2021-04-01:
Hi.
First of all thank you for for your effort producing this code. I have been running it on an ESP32 WROOM. I need to be able to determine whether the client (FileZilla) has connected to the server running on the ESP32 so that I can turn off other process running in the main loop. I am also finding that file transfer from FileZilla to the ESP32 hangs as FileZilla is waiting for a (connect/proceed?) message that never arrives when I use server close using FileZilla the transfer appears to procced weird. I have tried modifying the library code so that a flag is changed in the connect disconnect / routines, this int is declared as global in the main routine but??? the strategy has not worked.
Best,
Peter G.

xreef, 2021-04-02:
Hi thebluecrab,
I'd likes to extend the library with more FS (i publish an article on littlefs on esp32 and next FatFS) and I'm going to add some callback called when some event are fired.
Some event like used space change, connect disconnect and so on, but i need  some time.
I must find some time then I add the connection callback for first.

Bye Renzo

xreef, 2021-06-16:
Hi thebluecrab,
now support 2 call back to manage connection and transfer progress.
I must integrate It with examples etc. but here a little piece of code to start.

```cpp

void _callback(FtpOperation ftpOperation, unsigned int freeSpace, unsigned int totalSpace){
	Serial.print(">>>>>>>>>>>>>>> _callback " );
	Serial.print(ftpOperation);
	/* FTP_CONNECT,
	 * FTP_DISCONNECT,
	 * FTP_FREE_SPACE_CHANGE
	 */
	Serial.print(" ");
	Serial.print(freeSpace);
	Serial.print(" ");
	Serial.println(totalSpace);

	// freeSpace : totalSpace = x : 360

	freeSpacePieData(freeSpace, totalSpace);

	if (ftpOperation == FTP_CONNECT) connectedDisconnected(true);
	if (ftpOperation == FTP_DISCONNECT) connectedDisconnected(false);
};
void _transferCallback(FtpTransferOperation ftpOperation, const char* name, unsigned int transferredSize){
	Serial.print(">>>>>>>>>>>>>>> _transferCallback " );
	Serial.print(ftpOperation);
	/* FTP_UPLOAD_START = 0,
	 * FTP_UPLOAD = 1,
	 *
	 * FTP_DOWNLOAD_START = 2,
	 * FTP_DOWNLOAD = 3,
	 *
	 * FTP_TRANSFER_STOP = 4,
	 * FTP_DOWNLOAD_STOP = 4,
	 * FTP_UPLOAD_STOP = 4,
	 *
	 * FTP_TRANSFER_ERROR = 5,
	 * FTP_DOWNLOAD_ERROR = 5,
	 * FTP_UPLOAD_ERROR = 5
	 */
	Serial.print(" ");
	Serial.print(name);
	Serial.print(" ");
	Serial.println(transferredSize);

	(ftpOperation==FTP_UPLOAD || ftpOperation==FTP_DOWNLOAD)?transfer(true, ftpOperation==FTP_UPLOAD):transfer(false, false);

	fileTransfer(ftpOperation, name, transferredSize);
};

void setup()
{
	ftpSrv.setCallback(_callback);
	ftpSrv.setTransferCallback(_transferCallback);
```

if you need more info write here or better on my forum on www.mischianti.org .

Bye Renzo

closed 2021-06-16 by xreef