xreef/SimpleFTPServer #4: Error modifying files via ftp.

ForceConstant, 2021-11-18:
So I have this setup in my esp32 sketch, and am able to connect and view/upload files, but am not able to delete or rename files. 

For example I login and see /heart.gif, if I try to delete I get below.  

```
  --> FTP reconnected
Deleting file /heart.gif...
>>>  DELE /heart.gif
<<<  550 /heart.gif not found.

>>>  CWD .
<<<  257 "/" is your current directory
```

xreef, 2021-11-18:
Hi,
What configuration you use?
Bye Renzo

ForceConstant, 2021-11-18:
Setup just like the example in  the Readme.  Esp32 programmed with platform io.

xreef, 2021-11-18:
SPIFFS and wifi as standard?
Bye Renzo

ForceConstant, 2021-11-18:
Yes

xreef, 2021-11-19:
Hi ForceConstant,
I try and seems that all work correctly

[image]

can you give me some other information?

Bye Renzo

xreef, 2021-11-24:
No answer.

closed 2021-11-24 by xreef