xreef/SimpleFTPServer #5: No connection possible, always Error 421 Service not available

karstengit, 2021-11-28:
I tried to use your current version of the library, to build an FTP-Server with an ESP-12F in AP-mode.
I compiled it with #define FTP_SERVER_DEBUG, but there is no message what is happening.

The server starts and says "Ftp server waiting for connection on port 21" but no connection is possible.

It is possible to see the client for a moment before the connection is closed:
Client 192.168.4.101 Port 43866
Client 192.168.4.101 Port 43868
Client 192.168.4.101 Port 43870

I have no idea what can be done?

karstengit, 2021-11-28:
Now i tried a more basic example and the ftp answers but crashes.
First i tried a list command, but this fails.

```
FTP started!
 Ftp server waiting for connection on port 21
 Client connected!
FTP: Connected!
-U-S-E-R- -f-t-p-
-
Command is: USER
-P-A-S-S- -m-a-k-e-r-
-
Command is: PASS
 Authentication Ok. Waiting for commands.
-S-Y-S-T-
-
Command is: SYST
-P-O-R-T- -1-9-2-,-1-6-8-,-4-,-1-0-0-,-1-5-6-,-2-1-3-
-
Command is: PORT
 Data IP set to 192.168.4.100
 Data port set to 40149
-L-I-S-T-
-
Command is: LIST
List of file!!List Data stop!!-P-O-R-T- -1-9-2-,-1-6-8-,-4-,-1-0-0-,-2-1-4-,-5-
-
Command is: PORT
 Data IP set to 192.168.4.100
 Data port set to 54789
-S-T-O-R- -p-a-r-t-.-l-o-g-
-
Command is: STOR
CREATE FILE!!
File to open /part.log readType w+
TRUE
open/create 1
 Receiving part.log
FTP: Upload start!
NO DATA AVAILABLE!
NO DATA AVAILABLE!
NO DATA AVAILABLE!
NO DATA AVAILABLE!
NO DATA AVAILABLE!
NB -> 868
RC -> 0

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021bdb9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb20 end: 3fffffc0 offset: 0190
3ffffcb0:  3fff1f62 401056a7 3fff0adc 3ffefd84  
3ffffcc0:  00000000 00000000 3ffffd20 40218ed5  
3ffffcd0:  3ffeec30 402207b5 3fff0adc 3ffe83d0  
3ffffce0:  00000014 3ffe83d0 4021ea94 4021eb58  
3ffffcf0:  00000005 3ffe8948 00000014 3ffefde8  
3ffffd00:  3ffefde8 3fff1f2c 3fff1f2c 3ffe895e  
3ffffd10:  3ffe895d 00000000 3ffffdd0 4021efa9  
3ffffd20:  00000000 ffffffff 00000000 00000000  
3ffffd30:  00007fff 00000014 3f302073 3fff00a0  
3ffffd40:  3ffefde8 3ffefe20 3fff0094 40226b88  
3ffffd50:  00007fff 022c57ff 4010300e 00000100  
3ffffd60:  00000004 00000000 00000020 40100168  
3ffffd70:  4010221d 00000108 00000000 40101dec  
[... 159 lines cut ...]
	// Connect to Wi-Fi network with SSID and password
	Serial.println(F("Setting AP (Access Point) …"));
	// setting up a Wifi AccessPoint
	SetWifi("ESPAP","");

	/////FTP Setup, ensure SPIFFS is started before ftp;  /////////
#ifdef ESP32       //esp32 we send true to format spiffs if cannot mount
	if (SPIFFS.begin(true)) {
#elif defined ESP8266
	if (SPIFFS.begin()) {
#endif
		ftpSrv.setCallback(_callback);
		ftpSrv.setTransferCallback(_transferCallback);

		Serial.println(F("SPIFFS opened!"));
		ftpSrv.begin("ftp","maker");    //username, password for ftp.   (default 21, 50009 for PASV)
		Serial.println(F("FTP started!"));
	}    
	}

void loop(void){
	ftpSrv.handleFTP();        //make sure in loop you call handleFTP()!!  
	// server.handleClient();   //example if running a webserver you still need to call .handleClient();
}
´´´

xreef, 2021-11-28:
Hi,
the simple ftp server have some limitations, check the filezilla configuration on the article to do a correct connection.
Bye Renzo

karstengit, 2021-11-28:
The connection has been done with ftp in a shell.

Now it is running, but the reason is not clear.
Thanks for your support and library.

xreef, 2021-11-28:
Ahhh perfect!! 
If you need help write here or better on the site.
Bye Renzo

closed 2021-11-29 by xreef