xreef/SimpleFTPServer #7: Problem with Buffer size when compiling (esp32 Dev kit V1)

Gabryxx, 2021-12-18:
SimpleFTPServer/FtpServer.h:639:30: warning: 'packed' attribute ignored for field of type 'uint8_t [2048]' {aka 'unsigned char [2048]'} [-Wattributes]

xreef, 2021-12-19:
Hi Gabry,
I know that warning, but don't generate any problems.
Thanks Renzo

closed 2021-12-19 by xreef

Gabryxx, 2021-12-23:
Ok, ma il buffer viene generato coi valori immessi oppure no?
    Cioè se gli metto 1024 viene generato un buffer 1024 o viene
      ignorato il comando?
    ma mi rispondo da solo : Mi sembra che la velocità di
      trasferimento sia la stessa, per cui non viene considerato
    Se ti interessa sto sviluppando un esempio che usa anche la tua
      libreria e le LVGL con Esp32 e touch screen Ili9488
    qua il link ...guarda l'ultimo post che c'è il file zippato :
https://forum.lvgl.io/t/esp32-use-with-lvgl-library-need-a-little-example/7295/10
    Ciao
    
    
    Il 19/12/2021 09:11, Renzo Mischianti
      ha scritto:
    
    
      Hi Gabry,
        I know that warning, but don't generate any problems.
        Thanks Renzo
      —
        or Android.
        
      [
{
"target": "https://github.com/xreef/SimpleFTPServer/issues/7#issuecomment-997347767",
"url": "https://github.com/xreef/SimpleFTPServer/issues/7#issuecomment-997347767",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

xreef, 2021-12-23:
Ciao Gabry,
ahh ok la domanda perciò è se modificando FTP_BUF_SIZE cambia la dimensione del buffer.. 
La risposta è si, ma comunque sarà vincolato dalla libreria che effettua la lettura dei file e dalla dimensione del buffer della connessione.
Perciò magari metti 32000 ma non è supportato dai canali e rimane a 1024 che è lo standard in molti casi.
Se attivi il debug dovresti averne visione.

Con le LVGL ho iniziato pure io a fare qualche cosa, ma ora sono sotto con le librerie LoRa ed il cronotermostato è rimasto li senza vita.

Ciao Renzo

Gabryxx, 2021-12-24:
Mmmm....usi LoRa perchè il termotermostato è molto
      distante?....non l'ho provato ma le recensioni che si leggono in
      giro non sono così entusiasmanti da giustificare
      l'acquisto/aggiunta a un Esp32 col normale wifi e magari
      un'antenna direzionale esterna...qua un test delle antenne 360
      gradi, ma io ne ho una direzionale che aumenta di parecchio i Db:
    https://diyprojects.io/esp32-cam-solder-the-external-antenna-to-improve-the-range-and-stability-of-the-video-stream/#.YcWLtvDMJhE
    ciao e buon divertimento
    
    
    Il 23/12/2021 14:20, Renzo Mischianti
      ha scritto:
    
    
      Ciao Gabry,
        ahh ok la domanda perciò è se modificando FTP_BUF_SIZE cambia la
        dimensione del buffer..
        La risposta è si, ma comunque sarà vincolato dalla libreria che
        effettua la lettura dei file e dalla dimensione del buffer della
        connessione.
        Perciò magari metti 32000 ma non è supportato dai canali e
        rimane a 1024 che è lo standard in molti casi.
        Se attivi il debug dovresti averne visione.
      Con le LVGL ho iniziato pure io a fare qualche cosa,
        ma ora sono sotto con le librerie LoRa ed il cronotermostato è
        rimasto li senza vita.
      Ciao Renzo
      —
        or Android.
        
      [

{

"target": "https://github.com/xreef/SimpleFTPServer/issues/7#issuecomment-1000300568",

"url": "https://github.com/xreef/SimpleFTPServer/issues/7#issuecomment-1000300568",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

Gabryxx, 2021-12-24:
Per la precisione l'antenna che uso è questa: 
    
    D-Link ANT24-0600 Indoor 2.4GHz 6dBi Gain Antenna
    
    Il 24/12/2021 10:04, gabryxxxxp ha
      scritto:
    
    
      
      Mmmm....usi LoRa perchè il termotermostato è molto
        distante?....non l'ho provato ma le recensioni che si leggono in
        giro non sono così entusiasmanti da giustificare
        l'acquisto/aggiunta a un Esp32 col normale wifi e magari
        un'antenna direzionale esterna...qua un test delle antenne 360
        gradi, ma io ne ho una direzionale che aumenta di parecchio i
        Db:
      https://diyprojects.io/esp32-cam-solder-the-external-antenna-to-improve-the-range-and-stability-of-the-video-stream/#.YcWLtvDMJhE
      ciao e buon divertimento
      
      
      Il 23/12/2021 14:20, Renzo Mischianti
        ha scritto:
      
      
        Ciao Gabry,
          ahh ok la domanda perciò è se modificando FTP_BUF_SIZE cambia
          la dimensione del buffer..
          La risposta è si, ma comunque sarà vincolato dalla libreria
          che effettua la lettura dei file e dalla dimensione del buffer
          della connessione.
          Perciò magari metti 32000 ma non è supportato dai canali e
          rimane a 1024 che è lo standard in molti casi.
          Se attivi il debug dovresti averne visione.
        Con le LVGL ho iniziato pure io a fare qualche
          cosa, ma ora sono sotto con le librerie LoRa ed il
          cronotermostato è rimasto li senza vita.
        Ciao Renzo
        —
          or Android.
          
        [

{

"target": "https://github.com/xreef/SimpleFTPServer/issues/7#issuecomment-1000300568",

"url": "https://github.com/xreef/SimpleFTPServer/issues/7#issuecomment-1000300568",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

xreef, 2021-12-27:
No no il LoRa è un'altra attività.. 
Io uso i Tenda O3 per controllare i dispositivi a qualche chilometro di distanza.
Ciao Renzo