xreef/SimpleFTPServer #9: plz add anonymouse user and correct answer for browse in windows explorer

plaber, 2022-02-06:
Hi. In file ftpserver.cpp line 322 i just added
    if( ! strcmp( parameter, pass ) || user == "anonymous")
so i can config as
    ftpSrv.begin("anonymous","password");
and it allow to use anon connections, so then i opened ftp://192.168.0.123/ in windows explorer i see files as
+/,	12342,	01.bmp
+/,	12342,	09.bmp
etc..
can you add correct answer for LIST command for windows explorer? its much simplify using library without any ftp client program

plaber, 2022-02-06:
wow man, i make it works. for windows list answer format
<datetime>_space_<size>_space_<name>
so i change commands in list function to
		data.print( F("01-30-18 15:43PM ") );
		data.print( long( fileDir.size()) );
		data.print( F(" ") );
		data.println( fileDir.name() );
and windows explorer show files with correct size and date, awesome

plaber, 2022-02-06:
now i need to cutting filenames to 32 characters with extension to fit ffatfs requirements

xreef, 2022-02-06:
Hi @plaber,
It's very interesting.
I'll add support for anonymous user in the next week, and I will check how IE manage FTP.
Thanks Renzo

xreef, 2022-02-23:
Hi @plaber,
when you have some time check the development.
Bye Renzo

plaber, 2022-02-25:
browse in windows fine, now i cant understand how to use anon connection
i see you add second version of void FtpServer::begin( const char * _welcomeMessage )
but if i try to compile ftpSrv.begin("Welcome!"); it says call of overloaded 'begin(const char [9])' is ambiguous

plaber, 2022-02-25:
if user anonymous it must accept any password looks like email: anonymous@example.com or @test, but i dont see in PASS command block any changes to handle it

xreef, 2022-02-25:
Ops... ok I fixed It.
Thanks Renzo

plaber, 2022-02-26:
ok. now it works. is it possible to switch storage type without editing library files? it is not a good practice to edit, because every library update my changes are lost

plaber, 2022-02-26:
i trying something like this
`#define DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32 4
#define DEFAULT_STORAGE_TYPE_ESP32 8
#include <SimpleFTPServer.h>`
nothing happened

xreef, 2022-02-26:
If you want change from your file remove the FtpServerKey.h include and copy the content in your file.
Bye Renzo

plaber, 2022-04-10:
is it possible to add utf8 encoding for non ascii characers? if file name contains cyrillic symbols it brokes

xreef, 2022-04-10:
Hi @plaber,
can you give me, or attach a file to help the develop.
Bye Renzo

plaber, 2022-04-11:
just name some file for example Акылбай.txt and upload it by arduino ide spiffs, and when you open ftp you will see РђРєС‹Р»Р±Р°Р№.txt

plaber, 2022-04-24:
at 2.0.0 version ESP8266 doList send wrong file names, looks like fn variable contains trash

xreef, 2022-04-24:
Hi @plaber,
I try the latest version with esp8266 and LittleFS, and all seems to work correctly.
Bye Renzo

xreef, 2022-04-24:
The encoding, It's more difficult than expected. I must find a good quantity of time to find a solution.
bye Renzo

plaber, 2022-05-01:
in cpp file line 1470 this
```
const char* fn = dir.fileName().c_str(); Serial.printf("test %s ", fn)
```
show me rubbish
but if i add intermediate variable
```
String aza = dir.fileName();
const char* fn = aza.c_str(); Serial.printf("test %s ", fn)
```
this fix the problem

plaber, 2022-05-01:
at line 1382 you did like this, but at 1470 you not

xreef, 2022-05-01:
Hi @plaber,
thaks again for your support, I'm going to add your change to the library in the next release.

The encoding is a big problem.

Bye Renzo

plaber, 2022-05-02:
so, i am add this
```
  else if( CommandIs( "OPTS" )) {
    if( ParameterIs( "UTF8 ON" ) || ParameterIs( "utf8 on" )) {
      client.println(F("200 OK, UTF8 ON") );
      DEBUG_PRINT(F("200 OK, UTF8 ON") );
    } else {
      client.println(F("504 Unknow OPTS") );
	  DEBUG_PRINT(F("504 Unknow OPTS") );
    }
  }
```
to code and windows show me correct symbols, but now the problem is when i trying to copy or rename it brokes, server need to detect if utg8 command was getted and string contains character >128 to glue them to string by 2 bytes group

plaber, 2022-05-06:
if i disable legalchar cheking during rename it will work, copy also working, so its easy to add utf8 support, just add opts answer and change legalchar cheking function

plaber, 2022-05-06:
why rename is copy/remove function if FS contains rename function?

xreef, 2022-05-06:
Hi @plaber,
thanks for your precious work, I must find some time to upgrade the library.

I use this definition https://github.com/xreef/SimpleFTPServer/blob/master/FtpServer.cpp#L2366 to support rename in all situations, but I must do It more restrictive, rename not exist only for SD in Arduino UNO standard library.

Can you send me your code? I check It and modify the current version, and restart all the tests.

Thanks again, Bye Renzo

plaber, 2022-05-06:
i am send you code by vk.com

plaber, 2022-05-06:
another trick i used is to cut filename during upload
```
		String filename = upload.filename;
		int dot = filename.lastIndexOf(".");
		String name =  filename.substring(0, dot);
		String ext = filename.substring(dot);
		ext.toLowerCase();
		filename = name.substring(0,25) + ext;
```
you can think about this future, because file dissapear if i upload it to ftp and filename too long

plaber, 2022-05-07:
i saw your code, rename is ok, just my fail, no pretense

xreef, 2022-05-09:
> another trick i used is to cut filename during upload
>
> ```
> 		String filename = upload.filename;
> 		int dot = filename.lastIndexOf(".");
> 		String name =  filename.substring(0, dot);
> 		String ext = [...]

Also, the file name lenght isn't so simple to manage, not all filesystem has the same limitation, SD of Arduino UNO and SPIFFS before 2.4.2 esp8266 core have 8.3  LittleFS 32.3 and so on. To apply that management I must recheck all the filesystem limitations.

But probably I can do a limitation of 8.3 for SD Arduino and esp8266 before 2.4.2 and 32.3 for others.

Bye Renzo

xreef, 2022-05-17:
Release on version v2.1.0.
thanks Renzo

closed 2022-05-17 by xreef

plaber, 2022-05-21:
lol, you comitted Serial.printf at line 1363 https://github.com/xreef/SimpleFTPServer/blob/e8123a5e10b0053fae0691cd26b260f3618ae036/FtpServer.cpp#L1363

xreef, 2022-05-21:
Noooo... I fix It soon :'(

plaber, 2022-06-16:
can you plz add rename status to FtpOperation, so with setCallback function i will can catch any changes of files on my filesystem