xreef/SimpleFTPServer #11: Failed to compile examples for WIO terminal

gbernal, 2022-03-13:
Hi @xreef, first of all thank you for sharing this library. 

I just downloaded your library and tried to test it but I'm unable to compile with error.
Arduino version: 1.8.13
OS: Windows 10

```
exit status 1
Error compiling for board Seeeduino Wio Terminal.
```
I have have commented out the two define lines at the top
#define DEFAULT_FTP_SERVER_NETWORK_TYPE_SAMD NETWORK_SEEED_RTL8720DN
#define DEFAULT_STORAGE_TYPE_SAMD STORAGE_SDFAT2
 

and I get the following errors
`In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:0:
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:312:18: error: reference to 'File' is ambiguous
  #define FTP_DIR File
                  ^
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:495:20: note: in expansion of macro 'FTP_DIR'
   bool    openDir( FTP_DIR * pdir );
                    ^~~~~~~
In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:5:0:
C:\Users\gbern\Documents\Arduino\libraries\SdFat\src/SdFat.h:462:16: note: candidates are: typedef class FsFile File
 typedef FsFile File;
                ^~~~
In file included from C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:301:0,
                 from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:
C:\Users\gbern\Documents\Arduino\libraries\Seeed_Arduino_FS\src/Seeed_FS.h:57:11: note:                 class fs::File
     class File : public Stream {
           ^~~~
In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:0:
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:312:18: error: 'File' has not been declared
  #define FTP_DIR File
                  ^
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:495:20: note: in expansion of macro 'FTP_DIR'
   bool    openDir( FTP_DIR * pdir );
                    ^~~~~~~
In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:0:
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:508:15: error: expected ';' at end of member declaration
   const char* getFileName(FTP_FILE *file){
               ^~~~~~~~~~~
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:508:36: error: expected ')' before '*' token
   const char* getFileName(FTP_FILE *file){
                                    ^
In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:0:
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:311:19: error: reference to 'File' is ambiguous
  #define FTP_FILE File
                   ^
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:562:22: note: in expansion of macro 'FTP_FILE'
   uint16_t fileSize( FTP_FILE file );
                      ^~~~~~~~
In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:5:0:
C:\Users\gbern\Documents\Arduino\libraries\SdFat\src/SdFat.h:462:16: note: candidates are: typedef class FsFile File
 typedef FsFile File;
                ^~~~
In file included from C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:301:0,
                 from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:
C:\Users\gbern\Documents\Arduino\libraries\Seeed_Arduino_FS\src/Seeed_FS.h:57:11: note:                 class fs::File
     class File : public Stream {
           ^~~~
In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:0:
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:311:19: error: 'File' has not been declared
  #define FTP_FILE File
                   ^
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:562:22: note: in expansion of macro 'FTP_FILE'
   uint16_t fileSize( FTP_FILE file );
                      ^~~~~~~~
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:311:19: error: reference to 'File' is ambiguous
  #define FTP_FILE File
                   ^
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:637:3: note: in expansion of macro 'FTP_FILE'
   FTP_FILE     file;
   ^~~~~~~~
In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:5:0:
C:\Users\gbern\Documents\Arduino\libraries\SdFat\src/SdFat.h:462:16: note: candidates are: typedef class FsFile File
 typedef FsFile File;
                ^~~~
In file included from C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:301:0,
                 from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:
C:\Users\gbern\Documents\Arduino\libraries\Seeed_Arduino_FS\src/Seeed_FS.h:57:11: note:                 class fs::File
     class File : public Stream {
           ^~~~
In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:0:
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:312:18: error: reference to 'File' is ambiguous
  #define FTP_DIR File
                  ^
C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:638:3: note: in expansion of macro 'FTP_DIR'
   FTP_DIR      dir;
   ^~~~~~~
In file included from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:5:0:
C:\Users\gbern\Documents\Arduino\libraries\SdFat\src/SdFat.h:462:16: note: candidates are: typedef class FsFile File
 typedef FsFile File;
                ^~~~
In file included from C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer/FtpServer.h:301:0,
                 from C:\Users\gbern\AppData\Local\Temp\arduino_modified_sketch_30581\FTPServer_wio_terminal_SdFat.ino:9:
C:\Users\gbern\Documents\Arduino\libraries\Seeed_Arduino_FS\src/Seeed_FS.h:57:11: note:                 class fs::File
     class File : public Stream {
           ^~~~
`

```
Using library SdFat at version 2.1.2 in folder: C:\Users\gbern\Documents\Arduino\libraries\SdFat 
Using library SPI at version 1.0 in folder: C:\Users\gbern\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.2\libraries\SPI 
Using library Adafruit_ZeroDMA at version 1.0.4 in folder: C:\Users\gbern\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.2\libraries\Adafruit_ZeroDMA 
Using library Seeed_Arduino_rpcWiFi at version 1.0.5 in folder: C:\Users\gbern\Documents\Arduino\libraries\Seeed_Arduino_rpcWiFi 
Using library Seeed_Arduino_rpcUnified at version 2.1.3 in folder: C:\Users\gbern\Documents\Arduino\libraries\Seeed_Arduino_rpcUnified 
Using library Seeed_Arduino_FreeRTOS at version 1.1.0 in folder: C:\Users\gbern\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.2\libraries\Seeed_Arduino_FreeRTOS 
Using library SimpleFTPServer at version 1.2.1 in folder: C:\Users\gbern\Documents\Arduino\libraries\SimpleFTPServer 
Using library Seeed_Arduino_FS at version 2.1.1 in folder: C:\Users\gbern\Documents\Arduino\libraries\Seeed_Arduino_FS 
Using library Seeed_Arduino_mbedtls at version 3.0.1 in folder: C:\Users\gbern\Documents\Arduino\libraries\Seeed_Arduino_mbedtls 

```
I hope this helps, thank you in advance

gbernal, 2022-03-13:
I think the issue is that examples FTPServer_wio_terminal_SdFat and FTPServer_wio_terminal_SdFat_TFT_monitor as the file names say SDFat is creating an issue with Wio Terminal

xreef, 2022-03-14:
Hi @gbernal,
I think you import 2 SD libraries and this generates an issue.
You must select SDFat or Seeed_FS.
Bye Renzo

gbernal, 2022-03-16:
Thank you, I will go ahead and close this issue.

closed 2022-03-16 by gbernal