xreef/SimpleFTPServer #16: Add Teensy 4.1 Support

jakespeed1311, 2022-04-02:
Hello,
Please add Teensy 4.1 support. I think QN Ethernet is the better Network stack for ftp.

Also the  SdFat 2 and Littlefs on Teensy als filesystem would be great.

Tanks
Jake
PS: I trieyd but was not able to get a stable version.

xreef, 2022-04-02:
Hi Jake, 
I ask for a sample board from the PJRC; if the answer is ok, I'm happy to work on that.
Bye Renzo

xreef, 2022-04-11:
Hi @jakespeed1311,
Sorry, but no answer from https://www.pjrc.com/about/contact.html, so, for the support, you can try to contact them.
Bye Renzo

jakespeed1311, 2022-04-11:
Hi @xreef,
a borad + ethernet  costs today in germany 36€.  I don't know how much is it in Italy.
i don t think that pjrc will sponcer a borad now. 
Maybee we ask in the forum. But I don't not want that you only do the work ony  for me. 
So if no other people need that i do not want to waste your time. 
So i am not shure if you generaly interstet in Teensy?

Put thanks for have a try on that.
greedings
Jake

xreef, 2022-04-11:
Hi,
for the support of Arduino MKR Wifi 1010 (SAMD21 devices) Arduino send me a board, for LoRa E22 and LoRa E220 Ebyte send me a stock of devices.

I think It's an interest of PJRC that a set of libraries work with his devices, but I can't buy all devices even if there is a big request.

And Teensy isn't a device that I'd like to use in a project, for me It's expensive, so I don't has one of them.

I leave this issue open, in the future (now I'm exploring STM32) probably I buy one of them for studying purposes.

Bye Renzo

Electrolyt, 2022-04-18:
Hi @ xreef,
Hi @ jakespeed1311,

your FTP-Server-project is a very impressive work.
I'am also interessted for a solution of a working library on the teensy 4.1 board.
The teensy 4.1 is a realy fast microcomputer board for a basis of complex sketches.
I would be very happy if you could get excited about it and work for it.

Best greetings,

electrolyt.

closed 2022-05-26 by jakespeed1311

Electrolyt, 2022-05-27:
See link:
https://github.com/khoih-prog/FTP_Server_Teensy41