xreef/SimpleFTPServer #17: Filezilla returning Response:	425 No data connection

mjs513, 2022-05-08:
@xreef 
Need some help in figuring out why I keep getting the failure of the data connection.  

We are using a Teensy 4.1 primarily which we added support for in the library and after updating firmware and Adafruits WifiNINA for accept function managed to start seeing the connection to port 21.  But it keeps failing on the data connection.   Same thing occurs when we use a Adafruit SAMD21 Feather M0 board.

If we turn on debug messages we see:
```
NOT ANONYMOUS
esp8266
 Ftp server waiting for connection on port 21
 Client connected!
>>>>>>>>>>>>>>> _callback 0 1 1
CONNECTED
-U-S-E-R- -e-s-p-8-2-6-6-
-
Command is: USER
USER: esp8266 esp8266
-P-A-S-S- -e-s-p-8-2-6-6-
-
Command is: PASS
 Authentication Ok. Waiting for commands.
-S-Y-S-T-
-
Command is: SYST
-F-E-A-T-
-
Command is: FEAT
-P-W-D-
-
Command is: PWD
-T-Y-P-E- -I-
-
Command is: TYPE
-P-A-S-V-
-
Command is: PASV
 Connection management set to passive
 Listening at 192.168.1.239:50009
-M-L-S-D-
-
Command is: MLSD
List of file!!dataConnect failed!!Disconnect client!
 Disconnecting client
```
which pretty corresponds to what we see in filezilla.

```
Status:	Connection established, waiting for welcome message...
Status:	Plain FTP is insecure. Please switch to FTP over TLS.
Status:	Server does not support non-ASCII characters.
Status:	Logged in
Status:	Retrieving directory listing...
Command:	PWD
Response:	257 "/" is your current directory
Command:	TYPE I
Response:	200 TYPE is now 8-bit binary
Command:	PASV
Response:	227 Entering Passive Mode (192,168,1,239,195,89)
Command:	MLSD
Response:	425 No data connection
Error:	Failed to retrieve directory listing
```

Any help would be appreciated.

plaber, 2022-05-08:
do you check filesystem type used in library and in you code to be matched?

mjs513, 2022-05-08:
Was just looking at that now as I was converting from using SD to SDFat2.  Thing is looking at your openDir function I don't see a case for SdFat2 or SdFat1.  Havent finished the trace yet though

mjs513, 2022-05-08:
Ok a bit more debugging and its failing on this line:
`    if( dataConnect()){`
in function processCommand for command MLSD

plaber, 2022-05-09:
now you need to know what defined in FTP_CLIENT_NETWORK_CLASS

mjs513, 2022-05-09:
Hi @plaber 
Some one finally got it working last night by switching filezilla to active mode.  Tried it as well and it worked except now, at least for me, its failing to retrieve the directory in filezilla:
```
Command:	MLSD
Response:	150 Accepted data connection to port 50710
Response:	150 Accepted data connection to port 50710
Error:	Failed to retrieve directory listing
```
Not sure why yet though.  I switched to using the normal SD library.

EmileSpecialProducts, 2024-03-09:
This is the Filezilla that i use 
 
[image]

closed 2025-03-23 by xreef