xreef/SimpleFTPServer #18: The ftp server disconnects immediately

faustolnx, 2022-05-27:
Good morning,
I am trying your SimpleFTPServer library, but I have a problem.
I execute the command from cmd windows, ftp 10.0.0.177, the welcome message is displayed, but it disconnects immediately and displays the message 221 Goodbye, without giving me the possibility to insert user and password.
I am testing the library with an ESP32 and W5500 and it has been so configured

// esp32 configuration
#ifndef DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32
    #define DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32 NETWORK_W5100 // NETWORK_ESP32
    #define DEFAULT_STORAGE_TYPE_ESP32 STORAGE_FFAT
#endif

The LAN card works perfectly, I tried with http and udp examples.

Any help from you is appreciated in getting the library to work
Thank you
Fausto

[Test_SimpleFTP.zip](https://github.com/xreef/SimpleFTPServer/files/8785035/Test_SimpleFTP.zip)
[image]

xreef, 2022-05-27:
Hi,
I think the client stop and asking the user.

Use a GUI client like filezilla to test It.
Or resource explorer of windows.

Bye Renzo

dquadros, 2023-10-22:
I will create a more detailed issue, but this happens because the Windows FTP client sends an OPTS command before authorization and the server replies with a "530" error. Changing the server to accept the OPTS command even if not authorized yet solves this problem.

JiriBilek, 2024-07-16:
I also ran into this problem with windows10 ftp command. I think the solution is to move the code

https://github.com/xreef/SimpleFTPServer/blob/41cb655905553b9b858a58df2fce031f18d86e31/FtpServer.cpp#L707-L720

a few lines up - here:
https://github.com/xreef/SimpleFTPServer/blob/41cb655905553b9b858a58df2fce031f18d86e31/FtpServer.cpp#L424

Thanks for this library, it helped me a lot with my project.

closed 2025-01-11 by xreef