xreef/SimpleFTPServer #26: Read Time Out Error on MLSD via Android FTP Client apps like AndFTP, FTPCafe, etc.

outcastdreamer, 2022-07-09:
Hi, 
I am sorry to trouble you again. For my first test with SimpleFTPServer and windows FTP Client (Filezilla, WinSCP, etc), SimpleFTPServer works like a charm for ESP8266 with SD Card, but for Android FTP Client apps like AndFTP, FTPCafe, the same ESP8266 and SD Card connection gives *READ TIME OUT ERROR*. This is what the debug report shows :

```
 Ftp server waiting for connection on port 21
 Client connected!
-U-S-E-R- -u-s-e-r-
-
Command is: USER
USER: user user
-P-A-S-S- -p-a-s-s-w-o-r-d-
-
Command is: PASS
 Authentication Ok. Waiting for commands.
-F-E-A-T-
-
Command is: FEAT
-O-P-T-S- -U-T-F-8- -O-N-
-
Command is: OPTS
200 OK, UTF8 ON
-C-W-D- -/-
-
Command is: CWD
-T-Y-P-E- -A-
-
Command is: TYPE
-P-A-S-V-
-
Command is: PASV
 Connection management set to passive
 Listening at 192.168.193.148:50009
```
It just stops after this and the Client returns *READ TIME OUT ERROR* even after being on the same network.
I am assuming the MLSD command sent by the Android FTP Client is not parsed by the SimpleFTPServer maybe.

Really confused as to why this is happening. Let me know if you are able to connect to clients like "AndFTP" for SD Card with ESP8266.

Thank you for your time.

outcastdreamer, 2022-07-09:
I just noticed another thing, when I changed the connection from "passive" to **"active"** in AndFTP client settings for my FTP connection, then it gets connected but on passive it isn't working. I don't understand why it isn't working on passive connection for Android FTP Clients but works on passive connection with Windows FTP Client like Filezilla and WinSCP.

xreef, 2022-07-21:
Hi @outcastdreamer,
sorry for the late response, I test AndFTP and for me works correctly.
But witch version of the library do you have?
In the past, I fix a problem for esp8266 and SD.
Bye Renzo

closed 2022-07-28 by xreef