xreef/SimpleFTPServer #28: 553 File name not allowed. Too long.

Tri44, 2022-08-02:
i try to transfer file with name "IMG2022-07-25_13.44.30.jpg" and got "553 File name not allowed. Too long." error

can you able to explain why?

xreef, 2022-08-02:
Hi @Tri44,
can you give us some information about your test?
Bye Renzo

Tri44, 2022-08-02:
I try to download and upload file with name format like above i mention and got "553 File name not allowed. Too long." error

And i try to Up/Down load with short name "IMG20220210_131210.jpg" and its work
It just my file name just hit the allowed length of utf8 in this libraryi think?

xreef, 2022-08-02:
Yes it's probably, what microcontroller and storage do you use?

Tri44, 2022-08-02:
I use ESP32CAM and 4GB Micro SD from Vega
Also i change the
Remove <SPI.h>
Change <SD.h> to <SD_MMC.h>
STORAGE_MANAGER SD to STORAGE_MANAGER SD_MMC
In FtpServer.h

xreef, 2022-09-20:
Hi @Tri44,
I do multiple tests, and it seems that utf8 works correctly, I can't replicate it.
Bye Renzo

xreef, 2022-09-20:
Tested in the current release.

closed 2022-09-20 by xreef