xreef/SimpleFTPServer #30: Example Server: Disconnect Event never called

Joshy-2010, 2022-08-09:
Not a critical bug, but just wanted to report in case it helps anyone:
I used [this](https://www.mischianti.org/2020/02/08/ftp-server-on-esp8266-and-esp32/) example for a simple SPIFFS server based on esp32. Everything works fine, but after disconnecting filezilla 

case FTP_DISCONNECT:
      Serial.println(F("FTP: Disconnected!"));
      break;

is not called. At least I do not see the Serial log message. Strangely, all other messages (connect, file transfer, ...) are pritned correctly.

xreef, 2022-08-10:
Hi joshy,
I must verify. I don't know if the disconnection is from the client and the server does not detect It.
Bye Renzo

xreef, 2022-09-20:
Fixed in the next release.

closed 2022-09-20 by xreef