xreef/SimpleFTPServer #36: error: 'FTP_FILE_WRITE_APPEND' was not declared in this scope

h-atn, 2022-12-26:
Hi Renzo,
thanks for the great work, I could save any time and trouble.
I'm using the SimpleFTPServer-2.1.3 on a WeMos R1D1 with a SD-card-module
At first I dont the changes in the FtpServerKey.h the lines 
#ifndef DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP8266
     #define DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP8266 	NETWORK_ESP8266
    #define DEFAULT_STORAGE_TYPE_ESP8266 STORAGE_SD
After I had it done the compiler thrown errors: 'FTP_FILE_WRITE_APPEND' was not declared in this scope ...   and so on.
I found a bug/gap inside the FtpServer.h and  have it extented with same code like for STORAGE_SD_MMC
#elif(STORAGE_TYPE == STORAGE_SD)
	#include <SPI.h>
	#include <SD.h>
	#define STORAGE_MANAGER SD
  	#define FTP_FILE File
  	#define FTP_DIR File
	#define FTP_FILE_READ FILE_READ
	#define FTP_FILE_READ_ONLY FILE_READ
	#define FTP_FILE_READ_WRITE FILE_WRITE
// start of my extention
#ifdef ESP32
	#define FTP_FILE_READ_WRITE FILE_WRITE
	#define FTP_FILE_WRITE_APPEND FILE_APPEND
#else
	#define FTP_FILE_READ_WRITE FILE_WRITE
	#define FTP_FILE_WRITE_APPEND FILE_WRITE
#endif
	#define FTP_FILE_WRITE_CREATE FILE_WRITE
	#define FILENAME_LENGTH 32
// end of my extention    

Now it works fine.
Can it be that I have an old library used?
Otherwise I hope it helps others to take this hurdle.
Helmut

xreef, 2022-12-27:
Hi @h-atn,
you are right, I already push the fix on master, but I must release the new version on the Arduino library.
Thanks, I try to release it immediately.
Bye Renzo

xreef, 2022-12-27:
I release It. https://github.com/xreef/SimpleFTPServer/releases/tag/v2.1.4
Thanks again.

closed 2022-12-27 by xreef