xreef/SimpleFTPServer #38: ESP32 FireBeetle - No file access

SoilMaker, 2023-02-10:
Hi GitHub community
Thanks for this library, I'd love to use it for my project : a data-logger on ESP32 FireBeetle v.4 flash memory. I want an light and simple way to transfer the files from my ESP32 to my laptop so FTP seems to be a good protocol, more simple than a webserver. Arduino "WebServer" was working correctly and i could access SPIFFS without any trouble. 
I spent quite a lot of time this week trying to make SimpleFTPServer working but couldn't. I followed this tuto : https://www.mischianti.org/2020/02/08/ftp-server-on-esp8266-and-esp32/ and configure FileZilla properly. 
I first tried the [esp8266_esp32_SPIFFS] example, i can compile and transfer to my esp32 smoothly, but i can't see the files and when i try uploading i get disconnected. I thought it could be something weird with SPIFFS and moved to LittleFS, which seems to be better to SPIFFS anyway. However i have the same kind error message : 

> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1184 
> load:0x40078000,len:13104
> load:0x40080400,len:3036 
> entry 0x400805e4
> 
> ...
> Connected to SFR_EFA0
> IP address: 192.168.1.88
> LittleFS opened!
> FTP: Connected!
> [293131][E][vfs_api.cpp:24] open(): File system is not mounted
> [336161][E][vfs_api.cpp:24] open(): File system is not mounted
> [336162][E][vfs_api.cpp:24] open(): File system is not mounted
> FTP: Upload start!
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x4008ab39  PS      : 0x00060130  A0      : 0x80149590  A1      : 0x3ffb22d0
> A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000004  
> A10     : 0x3ffb8874  A11     : 0x80000001  A12     : 0x800905ac  A13     : 0x3ffb21c0
> A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000000  LBEG    : 0x4008ab39  LEND    : 0x4008ab49  LCOUNT  : 0xffffffff
> 
> 
> Backtrace: 0x4008ab36:0x3ffb22d0 0x4014958d:0x3ffb22e0 0x4014e126:0x3ffb25f0 0x4014e162:0x3ffb2680 0x400d8d42:0x3ffb26c0 0x400d2f52:0x3ffb2760 0x400d82fe:0x3ffb2780 0x400d85b5:0x3ffb27b0 0x400d3032:0x3ffb2800 0x400da231:0x3ffb2820
> 
> 
> 
> 
> ELF file SHA256: 240e2abd6ef5ded1
> 
> Rebooting...
> 

On FileZilla : 

> Status:	Logged in
> Status:	Retrieving directory listing...
> Status:	Directory listing of "/" successful
> Status:	Starting upload of C:\Users\Nicolas\Desktop\3_14H34.CSV
> Command:	PASV
> Response:	227 Entering Passive Mode (192,168,1,88,195,89)
> Command:	STOR 3_14H34.CSV
> Error:	Connection timed out after 20 seconds of inactivity
> Error:	File transfer failed

I also tried with WinSCP, it use to connect and send an error message "Can't acces file", right now it's just getting disconnected. 
I'm using platformIO but i also tried to upload from Arduino IDE 1.8 and 2.0.

> Last, I tried with an SD card but it's still not working : I get disconnected when i try to upload : 
> SD opened!
> >>>>>>>>>>>>>>> _callback 0 992313344 1003929600
> CONNECTED
> >>>>>>>>>>>>>>> _callback 1 992313344 1003929600
> DISCONNECTED

On FileZilla : 

> Status:	Connecting to 192.168.1.88:21...
> Status:	Connection established, waiting for welcome message...
> Response:	220---Welcome to Simply FTP server ---
> Response:	220---   By Renzo Mischianti   ---
> Response:	220 --    Version 2.1.6 (2023-02-02)    --
> Status:	Plain FTP is insecure. Please switch to FTP over TLS.
> Command:	USER user
> Response:	530 
> Error:	Could not connect to server

I briefly tried with other libraries (esp32_ftpclient ; esp8266ftpserver) but did not succeed. 

I googled a bit of everything tried to find other version of FireBeetle Boardmanager and so on... But I'm now stuck ! I would really appreciate some help. Thanks !

here is my code : 

`#include <WiFi.h>
#include <FS.h>
#include <LittleFS.h>

#include <SimpleFTPServer.h>

#define DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32     NETWORK_ESP32
#define DEFAULT_STORAGE_TYPE_ESP32 STORAGE_LITTLEFS

const char* ssid = "my ssid";
const char* password = "my pass";

FtpServer ftpSrv;   //set #define FTP_DEBUG in ESP8266FtpServer.h to see ftp verbose on serial

void _callback(FtpOperation ftpOperation, unsigned int freeSpace, unsigned int totalSpace){
  switch (ftpOperation) {
    case FTP_CONNECT:
      Serial.println(F("FTP: Connected!"));
      break;
    case FTP_DISCONNECT:
      Serial.println(F("FTP: Disconnected!"));
      break;
    case FTP_FREE_SPACE_CHANGE:
      Serial.printf("FTP: Free space change, free %u of %u!\n", freeSpace, totalSpace);
      break;
    default:
      break;
  }
};
void _transferCallback(FtpTransferOperation ftpOperation, const char* name, unsigned int transferredSize){
  switch (ftpOperation) {
    case FTP_UPLOAD_START:
      Serial.println(F("FTP: Upload start!"));
      break;
    case FTP_UPLOAD:
      Serial.printf("FTP: Upload of file %s byte %u\n", name, transferredSize);
      break;
    case FTP_TRANSFER_STOP:
      Serial.println(F("FTP: Finish transfer!"));
      break;
    case FTP_TRANSFER_ERROR:
      Serial.println(F("FTP: Transfer error!"));
      break;
    default:
      break;
  }

  /* FTP_UPLOAD_START = 0,
   * FTP_UPLOAD = 1,
   *
   * FTP_DOWNLOAD_START = 2,
   * FTP_DOWNLOAD = 3,
   *
   * FTP_TRANSFER_STOP = 4,
   * FTP_DOWNLOAD_STOP = 4,
   * FTP_UPLOAD_STOP = 4,
   *
   * FTP_TRANSFER_ERROR = 5,
   * FTP_DOWNLOAD_ERROR = 5,
   * FTP_UPLOAD_ERROR = 5
   */
   
};

void setup(void){
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  /////FTP Setup, ensure SPIFFS is started before ftp;  /////////

  /////FTP Setup, ensure SPIFFS is started before ftp;  /////////
  if (LittleFS.begin(true)) {

      ftpSrv.setCallback(_callback);
      ftpSrv.setTransferCallback(_transferCallback);

      Serial.println("LittleFS opened!");
      ftpSrv.begin("user","password");    //username, password for ftp.   (default 21, 50009 for PASV)
  }
}
void loop(void){
  ftpSrv.handleFTP();        //make sure in loop you call handleFTP()!!
}`

SoilMaker, 2023-02-27:
Thanks for your answer... and the solution ! It works !
I'm finally using a webserver for my project... definitely more complex, but it's nice to monitor and configure my sensor on the webpage.
Maybe I add a FTP connection on the project... just in case.

To solve this trouble I modified the FtpServerKey.h: 
`
// esp32 configuration
#ifndef DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32
	#define DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32 		NETWORK_ESP32
	#define DEFAULT_STORAGE_TYPE_ESP32 					STORAGE_FFAT
`
to
`#define DEFAULT_STORAGE_TYPE_ESP32 					STORAGE_LITTLEFS`

It was already defined in my sketch but i guess the library #define had the priority...

Bye.

closed 2023-02-27 by SoilMaker

xreef, 2023-02-27:
Perfect! thanks for your feedback.