xreef/SimpleFTPServer #39: upload to ftp server fails

pi-at-git, 2023-02-18:
Hi, I have a problem with the built-in example "esp8266_esp32_SPIFFS". While code compilation, upload to board, connection to my Wifi and connecting to the FTP-server from an Ubuntu client are fine for all, I get the same error as soon as I want to upload a file from my client: "552 Probably insufficient storage space"; on the serial monitor I see the following (followed by a board reboot): 

```
09:42:18.460 -> .....
09:42:20.489 -> Connected to service_net
09:42:20.522 -> IP address: 192.168.2.103
09:42:20.622 -> SPIFFS opened!
09:42:39.433 -> FTP: Connected!
09:42:49.272 -> FTP: Upload start!
09:42:49.338 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
09:42:49.405 -> 
09:42:49.405 -> Core  1 register dump:
09:42:49.438 -> PC      : 0x4008a991  PS      : 0x00060630  A0      : 0x8013a420  A1      : 0x3ffb22d0  
09:42:49.538 -> A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
09:42:49.637 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb2660  
09:42:49.704 -> A10     : 0x3ffc3e50  A11     : 0x0000029e  A12     : 0x0000001c  A13     : 0x00000002  
09:42:49.803 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
09:42:49.903 -> EXCVADDR: 0x00000000  LBEG    : 0x4008a991  LEND    : 0x4008a9a1  LCOUNT  : 0xffffffff  
09:42:50.003 -> 
09:42:50.003 -> 
09:42:50.003 -> Backtrace:0x4008a98e:0x3ffb22d00x4013a41d:0x3ffb22e0 0x4013efb6:0x3ffb25f0 0x4013eff2:0x3ffb2680 0x400d7a6a:0x3ffb26c0 0x400d2826:0x3ffb2760 0x400d7196:0x3ffb2780 0x400d744d:0x3ffb27b0 0x400d2906:0x3ffb2800 0x400d832d:0x3ffb2820 
```

Since I first thought, the board (WEMOS LOLIN32 Lite) might be damaged I checked SPIFFS: I can create a file directly in the Arduino IDE on the board. The board appears to be OK. I tried it out on two other boards (WEMOS D1 MINI ESP32, NODEMCU-32S). Outcome is exactly the same as for WEMOS LOLIN32 Lite. Do you have any idea what might be going on? Did I miss any instruction to pre-treat the flash?

pi-at-git, 2023-02-18:
I'm using the latest version of SimpleFTPServer (2.1.6) in the Arduino IDE.

pi-at-git, 2023-02-18:
Problem solved: FtpServerKey.h defines DEFAULT_STORAGE_TYPE_ESP32 as STORAGE_FFAT
replacing this line by

`#define DEFAULT_STORAGE_TYPE_ESP32 					STORAGE_SPIFFS`

does the job. Thanks for your great package!

closed 2023-02-18 by pi-at-git

xreef, 2023-02-20:
Hi pi-at-git,
perfect!
Bye Renzo