xreef/SimpleFTPServer #40: Some file upload correctly but end up being 0 in size

MrJake222, 2023-02-20:
As title says, filezilla reports successful transfer but files are replaced with 0 sized files. Uploading with override all files.

xreef, 2023-02-21:
Hi,
It's possible that there isn't enough space on the partition?
Bye Renzo

MrJake222, 2023-02-21:
No, it's almost empty, upload via serial works... (Whole filesystem)

xreef, 2023-02-21:
It's very strange, can you post the code/configuration and the devices you use?
Bye RM

MrJake222, 2023-02-22:
Does the server bind to specific interface or IP? My bootup sequence is:
* Open AP mode for 60 seconds, wait for connections
* **Here goes `FtpServer::begin`**
* If no station connects in said 60s, then go to station mode, disable AP mode
* Connect to configured AP

HTTP server works after this. FTP server refuses connections for passive mode, active mode connects but does this weird thing.

xreef, 2023-10-17:
Hi,
in this case i think you must set the local Ip because the device can't retrieve it, so if you do 

```cpp
      ftpSrv.begin("user", "password");    //username, password for ftp.  set ports in ESP8266FtpServer.h  (default 21, 50009 for PASV)

      IPAddress localIp = IPAddress();
      localIp.fromString("192.168.4.1");

      ftpSrv.setLocalIp(localIp);

```

it must work.
Sorry for the late response Renzo

closed 2023-10-17 by xreef