xreef/SimpleFTPServer #41: Flash memory usage?

maxbrito500, 2023-04-09:
Hello,

Even though there was ~10% of memory left to use on the ESP Wroom 32 device, when adding SimpleFTPServer this compilation issue comes up:
```
Linking .pio/build/esp32dev/firmware.elf
/home/brito/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/brito/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/brito/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/brito/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 2088 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```

This is how it looks without initializing the library (but including it nevertheless):
```
Linking .pio/build/esp32dev/firmware.elf
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  38.0% (used 124540 bytes from 327680 bytes)
Flash: [========= ]  87.9% (used 1152381 bytes from 1310720 bytes)
Building .pio/build/esp32dev/firmware.bin
esptool.py v4.4
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.

```

This should give around ~150KB of memory available to write. Is SimpleFTPServer really using more than that?

Is it possible to somehow reduce memory usage or make it fit?

Thank you for helping.

xreef, 2023-04-10:
Hi maxbrito,
the flash memory usage of an FTP server is minimal, but the library to manage an FTP like WiFi, WiFiClient, WiFiServer, SPIFFS, or LittleFS, or FFAT, needs to be installed.
But In general, if you use an esp32, you already preload that library to serve network requests so It's hidden.
Bye Renzo

maxbrito500, 2023-04-10:
Hello, thanks.

I already had those installed inside and compiled without issues. It is only when I'm initializing the FTPserver variable that this issue happens. I can't even declare the object for FTP that the memory went overflow.

closed 2023-08-28 by xreef