xreef/SimpleFTPServer #44: ESP32 SD storage doesn't use SD.rename method to rename a file

scuba-hacker, 2023-09-09:
**For the use case of ESP32 using SD storage: (particularly an issue for files that are very large, eg tens of MB like large wav files used in audio player use-case)**

In FTPServer.h the rename file method is not using the SD.rename method and instead the code makes a complete copy of the file, which is particularly an issue for large files as the copy method of rename takes a long time and hangs the FTP client, causing a disconnect. (Both Cyberduck and FileZilla).

I needed to make two changes:

**FTPServer.h**

Existing code in FTPServer.h:

#if STORAGE_TYPE == STORAGE_SD || STORAGE_TYPE == STORAGE_SD_MMC
  bool     rename( const char * path, const char * newpath );
#else
  bool     rename( const char * path, const char * newpath ) { return STORAGE_MANAGER.rename( path, newpath ); };
#endif`

Code change:

`#if (STORAGE_TYPE == STORAGE_SD || STORAGE_TYPE == STORAGE_SD_MMC)
	#if defined(ESP32))
		bool     rename( const char * path, const char * newpath ) { return STORAGE_MANAGER.rename( path, newpath ); };
	#else
		bool     rename( const char * path, const char * newpath );
	#endif
#else
  bool     rename( const char * path, const char * newpath ) { return STORAGE_MANAGER.rename( path, newpath ); };
#endif`

**FTPServer.cpp**

Existing code:

`#if (STORAGE_TYPE == STORAGE_SD ||  STORAGE_TYPE == STORAGE_SD_MMC)
  bool     FtpServer::rename( const char * path, const char * newpath ){
		FTP_FILE myFileIn = STORAGE_MANAGER.open(path, FILE_READ);
		FTP_FILE myFileOut = STORAGE_MANAGER.open(newpath, FILE_WRITE);`

Code change:

`#if !defined(ESP32) && (STORAGE_TYPE == STORAGE_SD ||  STORAGE_TYPE == STORAGE_SD_MMC)
  bool     FtpServer::rename( const char * path, const char * newpath ){
		FTP_FILE myFileIn = STORAGE_MANAGER.open(path, FILE_READ);
		FTP_FILE myFileOut = STORAGE_MANAGER.open(newpath, FILE_WRITE);`

xreef, 2023-10-17:
I add the change, thanks scuba.

closed 2023-10-17 by xreef

scuba-hacker, 2023-10-17:
Nice one, thanks @xreef for a great library. I find it the easiest way to get files uploaded to the Adafruit SPI Flash SD Card (which uses onboard flash and presents it as though it had a physical SD card).