xreef/SimpleFTPServer #45: ESP32 crash on upload to SD

haschmide, 2023-09-14:
I am using an SD card with an ESP32 - using default SPI but with custom SS pin.
I can view the cards content, download files, delete files and create new directories.
But as soon as I try to upload any file the ESP32 will crash:

19:55:06.068 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
19:55:06.143 -> 
19:55:06.143 -> Core  1 register dump:
19:55:06.143 -> PC      : 0x400d34b3  PS      : 0x00060130  A0      : 0x800d35bc  A1      : 0x3ffb1f50  
19:55:06.143 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb20f4  A5      : 0x3f4166e0  
19:55:06.143 -> A6      : 0x00000000  A7      : 0xff000000  A8      : 0x8015d2f0  A9      : 0x3ffb1f20  
19:55:06.143 -> A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffd49bc  
19:55:06.143 -> A14     : 0x00000000  A15     : 0x00000005  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
19:55:06.143 -> EXCVADDR: 0x00000010  LBEG    : 0x400ebaac  LEND    : 0x400ebaf2  LCOUNT  : 0x0000003c  
19:55:06.333 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:55:06.333 -> configsip: 0, SPIWP:0xee
19:55:06.333 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:55:06.370 -> mode:DIO, clock div:1
19:55:06.370 -> load:0x3fff0030,len:1184
19:55:06.370 -> load:0x40078000,len:13260
19:55:06.370 -> load:0x40080400,len:3028
19:55:06.370 -> entry 0x400805e4

haschmide, 2023-09-15:
Analysis shows that it crashes on data.flush() which in this case is ESP32 WiFiClient.flush() which seems to have a known issue of an possible endless loop in there forever: https://github.com/espressif/arduino-esp32/issues/4736

Suggest trying to workaround this...  for the combination here it works when removing data.flush() entirely

xreef, 2023-10-17:
Hi [haschmide](https://github.com/haschmide),
I confirm that it is the flush bug.
I tried to check in the lastest version, but I prefer that the bug will be fixed instead of doing a patch on the correct code.
Bye Renzo

xreef, 2023-10-17:
esp32 core 2.0.14 fixes the problem.
Thanks to all