xreef/SimpleFTPServer #46: issue with STORAGE_SDFAT2 with esp32 !!!

DR-DNK, 2023-10-07:
Hey ! I am using ESP32S3 WROOM2 N32R8V in Arduino IDE . I want to run SimpleFTPServer on esp32s3 with SdFat library  , so I can transfer my files directly to SD-Card . My SD-Card is initializing properly and I am able to connect with Filezilla . Also the content of my SD-Card is printed in the logs but my directories of SD-Card aren't getting listed in Filezilla thats why I am not able to transfer files .

Library verions :
SdFat                    2.2.2
SimpleFTPServer 2.1.6

Here's my arduino code in the zip file and the logs are provided below in logs.txt file .

file:///home/dnk126/Desktop/SdFat_with_SimpleFTPServer_in_Esp32s3/SdFat_with_SimpleFTPServer.zip
[logs.txt](https://github.com/xreef/SimpleFTPServer/files/12837957/logs.txt)

scuba-hacker, 2023-10-10:
I had a similar problem, this might work...

Try making this one line change to SimpleFTPServer/FtpServerKey.h

Change Line 63 from this:

#define DEFAULT_STORAGE_TYPE_ESP32 	STORAGE_FFAT

To this:

#define DEFAULT_STORAGE_TYPE_ESP32 	STORAGE_SD

DR-DNK, 2023-10-11:
Well I tried that , I added the headers as you said  and also changed the 63 line in the FtpServerKey.h .

#include <SimpleFTPServer.h>
#include <FTPServerKey.h>

#define DEFAULT_STORAGE_TYPE_ESP32 STORAGE_SD 

But the code went under Guru meditation as I tried to upload the file .
Can you please provide me with more insights ?

DR-DNK, 2023-10-13:
@scuba-hacker 
Hey!!! I tried that but that didnt work out as I got Guru Meditation error , can you please provide me with further suggestions ?

scuba-hacker, 2023-10-13:
Check that you're using passive FTP mode in Filezilla and max connections is set to 1. As you're getting that error I'd say undo the last change I suggested.

Another thing you could try is running the example provided with the library which uses SD.h, rather than SDFat.h. If it works it suggests that might be SDFat with an issue.

DR-DNK, 2023-10-14:
@scuba-hacker 
I tried that , but still I am not able to see the listing of the file on filezilla , also I am not able to transfer any file .
It is working with SD library , which is there in Arduino default example but I want to make it run with SdFat .

Let me provide you with some more insights . I have created the directories in my SD-Card with the default code provided in the SdFat libraries . Some SD-Card error is there I am facing don't know why , but still I am able to create the directory  successfully in my SD-Card as you can see in the logs .
Logs for that are provided below .
[DirectoryFunctions_logs.txt](https://github.com/xreef/SimpleFTPServer/files/12903134/DirectoryFunctions_logs.txt)

Than I have uploaded the code to connect with filezilla . 
Logs for that are provided below .
[FTP_CONNECTED_logs.txt](https://github.com/xreef/SimpleFTPServer/files/12903135/FTP_CONNECTED_logs.txt)

Then I am uploading the file from filezilla .
Logs for that are provided below .
[FILE_UPLOAD_LOGS.txt](https://github.com/xreef/SimpleFTPServer/files/12903128/FILE_UPLOAD_LOGS.txt)

Please guide me further .

scuba-hacker, 2023-10-14:
@DR-DNK Sorry, but I can't get SDFat (the QuickStart and SoftwareSPI examples) working on my ESP32-C3 dev board so can't help further. SD.h works fine for me for SimpleFTPServer with the change I described above.

DR-DNK, 2023-10-16:
@scuba-hacker 
Thanks @scuba-hacker for replying .

xreef, 2023-10-17:
Hi,
to use SdFat v.2.x you must set 
`#define DEFAULT_STORAGE_TYPE_ESP32 STORAGE_SDFAT2` not `STORAGE_SD`.
Bye Renzo

DR-DNK, 2023-10-18:
@xreef 
Have you tried using SdFat with esp32 for SimpleFTPServer .
The issue is that everything is working fine except my Files are not getting listed on FIlezilla .

DR-DNK, 2023-10-20:
Hey @xreef !!!

I've checked the core version, and as you suggested, I've updated it to 2.0.14. However, I'm still encountering the same error that @haschmide experienced. My directories are being listed from SD-Card on filezilla , but when I attempt to upload a file, it triggers a Guru Meditation error, causing my controller to reboot.

There's one more thing , as I have told you before , when I am using SD-Card with 512 MB storage , my directories are still not getting listed . So I tested with 4 GB storage , in which I was successful with listing of the directories on filezilla.

I am providing you with my code file and the log file which consist of the output and the error I am encountering with 4 GB SD-Card .
Please help me to make SdFat run with SimpleFTPServer to transfer files .

My FTPServerKey configurations for esp32 are ,
// esp32 configuration
#ifndef DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32
	#define DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32 		NETWORK_ESP32
	#define DEFAULT_STORAGE_TYPE_ESP32 					STORAGE_SDFAT2

[esp32_SdFat_SimpleFTPServer.zip](https://github.com/xreef/SimpleFTPServer/files/13053703/esp32_SdFat_SimpleFTPServer.zip)
[error_logs.txt](https://github.com/xreef/SimpleFTPServer/files/13053705/error_logs.txt)

DR-DNK, 2023-10-23:
Hey @xreef  !!!
Do you have any insights on my previous issue ?
I am still facing the same issue (Guru Meditation error) even after upgrading the core to 2.0.14 . 
Please help me out here .

DR-DNK, 2023-10-26:
@xreef and @scuba-hacker 
Thanks a lot guys , everything is working fine now . There was a little problem with my code that was causing the error .
SimpleFTPServer is a great library @xreef .

Conclusion : SdFat is working fine with esp32 which is connected to SD-Card while using SimpleFTPServer .
// esp32 configuration
#ifndef DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32
	#define DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32 		NETWORK_ESP32
	#define DEFAULT_STORAGE_TYPE_ESP32 					STORAGE_SDFAT2
	/**