xreef/SimpleFTPServer #55: Library SD Example not working with Adafruit Feather esp32 V2

nitin710, 2024-01-24:
Hi @xreef, thanks for the brilliant library! I'm so glad to find this in my google-ing!!
I think this solves what I'm trying to do but for some reason I cannot get it to work with my [Adafruit ESP32 V2](https://www.adafruit.com/product/5400) board.

### Here is what I'm trying to do:
1. I have an SD card attached to the SPI lines of the Feather.
2. I am trying to create a FTP server on the ESP32 so that I can access the data on the card without having to plug the SD Card out.

### Here is what I have tried:
1. I presume the [Arduino_esp32_SD.ino](https://github.com/xreef/SimpleFTPServer/blob/master/examples/Arduino_esp32_SD/Arduino_esp32_SD.ino) is the example I want to explore, so I tried building that for my board.
2. I changed the SPI pins to correctly reflect the SPI pins on my board.

### Things that are working:
1. I can build the code and upload it to my board.
2. I have verified that the FTP server is created. (using filezilla as the client)
    - The Serial prints "CONNECTED" and "DISCONNECTED" when I am working with filezilla, so I think the server acknowledges the connection. 
4. And to verify the SD is detected, I added `SD.ls()` commands in the [_callback](https://github.com/xreef/SimpleFTPServer/blob/11091f8ed341b363df06ea3c102fec460034037d/examples/Arduino_esp32_SD/Arduino_esp32_SD.ino#L21). It is indeed working (prints the contents of the SD card accurately).

### Here is where I stuck:
1. The SD file system does not show up on the FTP server.
    - All I see on the FTP server is the root folder `/`

### What i think may be going wrong
- Looking through the code, i think one of the `#defines` is not correct (maybe)
- Particularly,         [#define DEFAULT_STORAGE_TYPE_ESP32                                      STORAGE_FFAT](https://github.com/xreef/SimpleFTPServer/blob/11091f8ed341b363df06ea3c102fec460034037d/FtpServerKey.h#L63) should maybe
  -   `#define DEFAULT_STORAGE_TYPE_ESP32                                      STORAGE_SD`?
- It tried that change but did not solve the issue. I'm thinking I'm missing some other `#define`. 

Not clear what the issue may be, but any help is appreciated.

nitin710, 2024-01-24:
So I just uninstalled the Esp32 core.

I downgraded it to v2.0.1, to see if i can get this to work on my [Adafruit Feather Huzzah](https://www.adafruit.com/product/3619), and the FTP server started showing the SD card contents on the server.

I then ungraded the core to v2.0.13 (what i was using before), built the project for esp32-V2, and viola! Its working now.

Only change was 
`#define DEFAULT_STORAGE_TYPE_ESP32 					STORAGE_SD`

If the FTP server settings and correctly set as shown [here](https://mischianti.org/ftp-server-on-esp8266-and-esp32/), then i can download the files from the server!

However, I still cannot upload files to the server.

### Here is the log from the server
```
Status:	Connecting to 192.168.1.187:21...
Status:	Connection established, waiting for welcome message...
Status:	Plain FTP is insecure. Please switch to FTP over TLS.
Status:	Logged in
Status:	Starting upload of D:\Downloads\test-1.txt
Status:	Retrieving directory listing of "/"...
Command:	TYPE I
Response:	200 TYPE is now 8-bit binary
Command:	PASV
Response:	227 Entering Passive Mode (192,168,1,187,195,89)
Command:	MLSD
Response:	150 Accepted data connection to port 50009
Response:	226-options: -a -l
Response:	226 13 matches total
Command:	TYPE A
Response:	200 TYPE is now ASCII
Command:	PASV
Response:	227 Entering Passive Mode (192,168,1,187,195,89)
Command:	STOR test-1.txt
Error:	Disconnected from server: ECONNABORTED - Connection aborted
Error:	File transfer failed
Status:	Connecting to 192.168.1.187:21...
Status:	Connection established, waiting for welcome message...
Status:	Plain FTP is insecure. Please switch to FTP over TLS.
Status:	Logged in
Status:	Starting upload of D:\Downloads\test-1.txt
Status:	Retrieving directory listing of "/"...
Command:	TYPE I
Response:	200 TYPE is now 8-bit binary
Command:	PASV
Response:	227 Entering Passive Mode (192,168,1,187,195,89)
Command:	MLSD
Response:	150 Accepted data connection to port 50009
Response:	226-options: -a -l
Response:	226 13 matches total
Command:	TYPE A
Response:	200 TYPE is now ASCII
Command:	PASV
Response:	227 Entering Passive Mode (192,168,1,187,195,89)
Command:	STOR test-1.txt
Error:	Disconnected from server: ECONNABORTED - Connection aborted
Error:	File transfer failed
Status:	Connecting to 192.168.1.187:21...
Status:	Connection established, waiting for welcome message...
Status:	Plain FTP is insecure. Please switch to FTP over TLS.
Status:	Logged in
Status:	Starting upload of D:\Downloads\test-1.txt
Status:	Retrieving directory listing of "/"...
Command:	TYPE I
Response:	200 TYPE is now 8-bit binary
Command:	PASV
Response:	227 Entering Passive Mode (192,168,1,187,195,89)
Command:	MLSD
Response:	150 Accepted data connection to port 50009
Response:	226-options: -a -l
Response:	226 13 matches total
Command:	TYPE A
Response:	200 TYPE is now ASCII
Command:	PASV
Response:	227 Entering Passive Mode (192,168,1,187,195,89)
Command:	STOR test-1.txt
Error:	Disconnected from server: ECONNABORTED - Connection aborted
Error:	File transfer failed
```

### Here is the serial monitor
```
IP address: 192.168.1.187
SD opened!
Listing directory: /
  FILE: config.txt  SIZE: 71
  FILE: foo.txt  SIZE: 13
  FILE: 2023-08-02_23-20-43-877847_info.json  SIZE: 2806
  FILE: 2023-08-02_23-20-43-877847.csv  SIZE: 152788
  FILE: 2023-08-03_15-55-59-463493_info.json  SIZE: 2806
  FILE: 2023-08-03_15-55-59-463493.csv  SIZE: 44283735
  FILE: 2023-10-31_10-32-34-693976_info.json  SIZE: 2806
  FILE: 2023-10-31_10-32-34-693976.csv  SIZE: 3896594
  FILE: 2023-10-31_11-01-25-730647_info.json  SIZE: 2806
  FILE: 2023-10-31_11-01-25-730647.csv  SIZE: 537988
  FILE: 2024-01-09_10-07-37-244558_info.json  SIZE: 2806
  FILE: 2024-01-09_10-07-37-244558.csv  SIZE: 48099
  DIR : System Volume Information
>>>>>>>>>>>>>>> _callback 0 1 1
CONNECTED
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d32a7  PS      : 0x00060130  A0      : 0x800d33c8  A1      : 0x3ffb1f50  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb20f4  A5      : 0x3f412008  
A6      : 0x00000000  A7      : 0xff000000  A8      : 0x801588a0  A9      : 0x3ffb1f20  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffcd860  
A14     : 0x00000000  A15     : 0x00000005  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x400e5c28  LEND    : 0x400e5c6e  LCOUNT  : 0x0000003c  

Backtrace: 0x400d32a4:0x3ffb1f50 0x400d33c5:0x3ffb1f80 0x400d87b2:0x3ffb1fa0 0x400d908e:0x3ffb2220 0x400d2d62:0x3ffb2270 0x400dbd61:0x3ffb2290

ELF file SHA256: 3dcbbb89eab6e746

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4

...
Connected to **** [WiFi Name]
IP address: 192.168.1.187
SD opened!
```
Looks like uploading the file is crashing the core. maybe a un-handled null-ptr exception? I'll dig around to see if i can find what is causing it.

xreef, 2024-01-25:
Hi nitin,
you can find some basic information [here](https://mischianti.org/ftp-server-on-esp8266-and-esp32/).

Yes, you can set the correct FS type (and network if you need to change).
Bye Renzo

closed 2024-02-06 by xreef