xreef/SimpleFTPServer #57: ESP32 Devkit V1 - SimpleFTPServer plus Ethernet W5500 plus Blynk

AAKFISHERMANN, 2024-02-19:
HI,
I have a project using ESP32 DevKit V1 with Wiznet W5500 attached (SPI) plus Blynk.
Using VScode and Platformio code doesn't compiles with this library attached (SimpleFTPServer.h).
I got a lot of errors regarding WiFi and Ethernet...
error: cannot declare field 'FtpServer::ftpServer' to be of abstract type 'WiFiServer'
error: redefinition of 'class UIPServer'
error: redefinition of 'class UIPUDP'
Using these libraries:
#include <Arduino.h>
#include <SPI.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <FS.h>
#include <SD.h>
#include <Wire.h>
#include <RTClib.h>
#include <HardwareSerial.h>
#include <EEPROM.h>
#include <NTPClient.h>
#include <BlynkSimpleEthernet.h>
#include <SimpleFTPServer.h>

AAKFISHERMANN, 2024-02-19:
More one information:
I tried these config lines on FtpServerKey.h:
#define NETWORK_W5100 			(4)		// Standard Arduino Ethernet library
#define NETWORK_ETHERNET		(4)		// Standard Arduino Ethernet library
WiFi error codes solved... But not ethernet errors

xreef, 2024-02-19:
Please paste all FtpServerKey.h
and your ino file.
Thanks

AAKFISHERMANN, 2024-02-20:
One question about SPIFFS version: I must have a partition with ota_0 , ota_1 and spiffs ok? spiffs partition must have space to allocate firmware.bin file right? Thanks

xreef, 2024-02-24:
Hi,
Yes there isn't any problem for that.
Bye Renzo

closed 2024-06-08 by xreef