xreef/SimpleFTPServer #65: ftplib can't access the root folder in ESP32/ESP32C3

firminxu, 2024-07-19:
Hello!
Firstly, thanks for the the your good job, I can build my ftp sever easily.
I have used version 2.1.7 with my ESP32/ESP32C3 device in PlatformIO environment for a while, it works well.
After update it to 2.1.8, I can't access root folder of SD. after I rolled back to 2.1.7, it works again.
I tested it with below python code.
`from ftplib import FTP
def download_and_delete_log_csv_files(host, user, password, remote_dir, local_dir):
    with FTP(host) as ftp:
        try:
            # Attempt to login to the FTP server
            ftp.login(user=user, passwd=password)
            print("Login successful.")
            # Change the current working directory on the server
            ftp.cwd(remote_dir)
            print(f"Changed working directory to {remote_dir}.")

            # Explicitly set passive mode
            ftp.set_pasv(True)

            # Get a list of all files in the remote directory
            files = []
            ftp.retrlines('LIST', lambda line: files.append(line))

            # Print all files in the directory
            print("Files in the remote directory:")
            for file_info in files:
                filename = file_info.split(None, 8)[-1]  # Simplify extracting the filename
                print(filename)

        except Exception as e:
            # Handle any exceptions that occur during the process
            print(f"An error occurred: {e}")

# Example usage

host = '192.168.229.145'
user = 'esp8266'
password = 'esp8266'
remote_dir = '/'  # Directory on the FTP server
local_dir = r'D:\temp'  # Local directory

# Call the function to download and delete log CSV files
download_and_delete_log_csv_files(host, user, password, remote_dir, local_dir)
`

xreef, 2024-07-20:
Hi firminxu,
do you have some log to send me?
Thanks Renzo

firminxu, 2024-07-23:
below is my code in PlatformIO, it is similiar to example,

firminxu, 2024-07-23:
When it works in version 2.1.8.
platformio.ini is as below:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = xreef/SimpleFTPServer@2.1.8
```
when I run python code I posted, I got 530 error
```
Login successful.
Changed working directory to /.
An error occurred: 530 
```

amgrays, 2024-08-11:
I have a similar issue - ESP8266/Arduino 2.3.2 - using FileZilla it sometimes connects and gets the directory once and after that error message as follows:

`Status:	Connecting to 192.168.51.47:21...
Status:	Connection established, waiting for welcome message...
Status:	Plain FTP is insecure. Please switch to FTP over TLS.
Status:	Server does not support non-ASCII characters.
Status:	Logged in
Status:	Retrieving directory listing...
Command:	PWD
Response:	211-Extensions supported:
Response:	 MLST type*;modify*;size*;
Response:	 MLSD
Response:	 MDTM
Response:	 MFMT
Response:	 UTF8
Response:	 SIZE
Response:	 SITE FREE
Response:	211 End.
Error:	Failed to parse returned path.
Error:	Failed to retrieve directory listing
`
Reverting to previous version  (2.1.7) resolves the problem.

vortigont, 2024-08-28:
same here with Arduino core 3.x, esp32-S3, esp32-c3 and LittleFS

esp log show this
```
[ 12709][E][vfs_api.cpp:23] open(): File system is not mounted
```

FTP client shows this
```
Status:	Logged in
Status:	Retrieving directory listing of "/"...
Command:	CWD /
Response:	550 / not found.
Error:	Failed to retrieve directory listing
Status:	Retrieving directory listing of "/"...
Command:	CWD /
Response:	550 / not found.
Error:	Failed to retrieve directory listing
Status:	Retrieving directory listing of "/"...
Command:	CWD /
```

ifrew, 2024-09-07:
I have had the same issue. I had thought it was because my ftpserver runs in a separate task, but building the ardunio_esp32_sd example I got the same issue and also the same mounting error. Looking through the ftpserverkey.h file I found the problem. The default store for esp32 in version 2.1.8 is STORAGE_FFAT and not STORAGE_SD. Simply change the default storage type to STORAGE_SD and you should be good.

Relevant section from ftpserverkey.h file in the simpleftpserver library source code as follows:

// esp32 configuration
#ifndef DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32
	#define DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32 		NETWORK_ESP32
	#define DEFAULT_STORAGE_TYPE_ESP32 					STORAGE_SD // STORAGE_FFAT
	/**

// esp32 configuration
#ifndef DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32
	#define DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32 		NETWORK_ESP32
	#define DEFAULT_STORAGE_TYPE_ESP32 					STORAGE_SD // STORAGE_FFAT
	/**

closed 2025-01-11 by xreef