xreef/SimpleFTPServer #72: Problem using STOR/RETR on an Esp32/SD board

gareth3lee, 2024-12-02:
There seems to be a problem using SimpleFTPServer to communicate with FileZilla.

I have uploaded a minimal stand alone SimpleFTPServer 2.1.8 (2024-07-17) to an ESP32 module and configured it to use STORAGE_SD and also defined FTP_SERVER_DEBUG and FTP_ADDITIONAL_DEBUG.  I have attached my Arduino code, but my configuration is not unconventional.

[TestEspFtp.ino.txt](https://github.com/user-attachments/files/17971480/TestEspFtp.ino.txt)

I am trying to transfer files from a Linux Mint host using FileZilla 3.58.0.  FileZilla will connect and show me a valid directory listing from the SD card.  I can move in and out of subdirectories on the SD card without any problem.

But when I try to upload or download files to the SD card FileZilla stalls, displays the message below, then after three attempts the transfer fails.  The ESP32 does not display any error at this point despite all the DEBUG options being active.  The connection just stalls.

> Status:	Directory listing of "/" successful
> Status:	Connecting to 10.1.1.15:21...
> Status:	Connection established, waiting for welcome message...

**However, after the connection has stalled, if I instruct FileZilla to disconnect from the server, as part of its shutdown the upload/download is unblocked and completes**.  I then get a string of diagnostics from the ESP as you would expect associated with the transfer and housekeeping functions.

>  FTP CONNECTED
>   Command Old: 2  Transfer Old: 0  Data Old: 0
>   Command    : 3  Transfer    : 0  Data    : 0
> -A-U-T-H- -T-L-S-
> -
> Command is: AUTH
> -A-U-T-H- -S-S-L-
> -
> Command is: AUTH
> -U-S-E-R- -f-t-p-
> -
> Command is: USER
> USER: ftp ftp
>   Command Old: 3  Transfer Old: 0  Data Old: 0
>   Command    : 4  Transfer    : 0  Data    : 0
> -P-A-S-S- -X-X-X-X-
> -
> Command is: PASS
> PASS: XXXX
> PASS PARAM: XXXX
> PASS OK: 0
>  Authentication Ok. Waiting for commands.
>   Command Old: 4  Transfer Old: 0  Data Old: 0
>   Command    : 5  Transfer    : 0  Data    : 0
> -O-P-T-S- -U-T-F-8- -O-N-
> -
> Command is: OPTS
> 200 OK, UTF8 ON
> -C-W-D- -/-
> -
> Command is: CWD
> -P-W-D-
> -
> Command is: PWD
> -T-Y-P-E- -A-
> -
> Command is: TYPE
> -P-A-S-V-
> -
> Command is: PASV
>  IP: 10.1.1.15
>  Soft IP: 10.1.1.15
>  Connection management set to passive
>  Listening at 10.1.1.15:50009
>   Command Old: 5  Transfer Old: 0  Data Old: 0
>   Command    : 5  Transfer    : 0  Data    : 1
> -R-E-T-R- -f-o-o-.-t-x-t-
> -
> Command is: RETR
> utf8_strlen8
> PATH --> /foo.txt ...EXIST!
> File to open /foo.txt readType r
> TRUE
>  Sending foo.txt size 13
> FTP ftp_transfer 2 foo.txt 13
> NB --> 13
> FTP ftp_transfer 3 foo.txt 13
>   Command Old: 5  Transfer Old: 0  Data Old: 1
>   Command    : 5  Transfer    : 1  Data    : 1
>  Transfer completed in 15 ms, 
> 0 kbytes/s
> FTP ftp_transfer 4 foo.txt 13
>   Command Old: 5  Transfer Old: 1  Data Old: 1
>   Command    : 5  Transfer    : 0  Data    : 1
> FTP ftp_callback 1 1 1
> FTP DISCONNECTED
>   Command Old: 5  Transfer Old: 0  Data Old: 1
>   Command    : 1  Transfer    : 0  Data    : 1
>  Ftp server waiting for connection on port 21
>   Command Old: 1  Transfer Old: 0  Data Old: 1
>   Command    : 2  Transfer    : 0  Data    : 0

Can anyone suggest how I might diagnose this problem?  Is there a limit on the number of TCP connections the ESP32 protocol stack can service as this is a PASV transfer, but the server should only be listening on ports 21 and 50009.  (I was unable to even get directories to list when I set FileZilla to operate in active mode).

Much appreciated,

Gareth

gareth3lee, 2024-12-03:
As an update I also tested the server with the command line linux ftp client (it appears to be tnftp) and experienced no problems suggesting this is due to the exact sequence of commands issued by FileZilla.

I also reactivated the callback functions in my code and recorded the last diagnostic messages from ESP32 before it stalls:

```
 DIR MLSD 
 DIR NEXT 
 Type=dir;Modify=19700101000000;Size=0; mqtt
 Connected!!
 DIR MLSD 
 All file read!!
   Command Old: 5  Transfer Old: 5  Data Old: 1
   Command    : 5  Transfer    : 0  Data    : 1
```

After the stall (when I manually instruct FileZilla to disconnect) the messages are:

```
FTP ftp_callback 1 1 1
FTP DISCONNECTED
  Command Old: 5  Transfer Old: 0  Data Old: 1
  Command    : 1  Transfer    : 0  Data    : 1
Ftp server waiting for connection on port 21
  Command Old: 1  Transfer Old: 0  Data Old: 1
  Command    : 2  Transfer    : 0  Data    : 0
 Client connected!
FTP ftp_callback 0 1 1
FTP CONNECTED
  Command Old: 2  Transfer Old: 0  Data Old: 0
  Command    : 3  Transfer    : 0  Data    : 0
-A-U-T-H- -T-L-S-
-
Command is: AUTH
```

It then continues the login sequence and transfers the file without incident.

Now that I know that the server operates correctly with other FTP clients I will continue my project and bypass FileZilla.  (I will probably create a customised download client using the FTP libraries in Python.)

However, I'm happy run some tests if there anyone is seeking any more information about the problem.

Thanks,

Gareth

mcclellansws, 2024-12-17:
I might have recently found and implemented a workaround for what could related to your issue. In my case, the client is python 2.7 ftplib script (legacy embedded sw) that issues PASV followed by STOR. I noticed in the wireshark packet trace that the SimpleFTPServer's response to the PASV was truncated and did not include the IP and PORT. 

My workaround was to issue the PASV response as a single string https://github.com/mcclellansws/SimpleFTPServer/commit/91ae9ef74f1636da1c091a82240b840c72dfae03 to minimize the chance the response was split across IP packets. My theory is the root bug is on the client side but for my immediate needs this server side tweak seems to be working.

gareth3lee, 2024-12-20:
Thanks for your post SteveM.  I grafted your patch into the source code and cleared the caches to ensure that it recompiled the updated source code into the build.  Alas, it appears to fail in the same way as before.  fileZilla displays the following message on its console:
```
Status:	Retrieving directory listing of "/mqtt/bw6168"...
Status:	Directory listing of "/mqtt/bw6168" successful
Status:	Retrieving directory listing of "/mqtt/bw6168/garage"...
Status:	Directory listing of "/mqtt/bw6168/garage" successful
Status:	Retrieving directory listing of "/mqtt/bw6168/garage/trina"...
Status:	Directory listing of "/mqtt/bw6168/garage/trina" successful
Status:	Retrieving directory listing of "/mqtt/bw6168/garage/trina/W"...
Status:	Directory listing of "/mqtt/bw6168/garage/trina/W" successful
Status:	Connecting to 10.1.1.15:21...
Error:	Connection timed out after 20 seconds of inactivity
Error:	Could not connect to server
Status:	Disconnected from server
Status:	Delaying connection for 5 seconds due to previously failed connection attempt...
Status:	Connecting to 10.1.1.15:21...
Error:	Connection timed out after 20 seconds of inactivity
Error:	Could not connect to server
Status:	Disconnected from server
Status:	Delaying connection for 5 seconds due to previously failed connection attempt...
Status:	Connecting to 10.1.1.15:21...
Error:	Connection timed out after 20 seconds of inactivity
Error:	Could not connect to server
```
As you can see I'm able to move through the directories without error but SimpleFTPServer stalls on the file transfer.

Once the timeout periods are complete the server is still functional which is why I am referring to it as a stall rather than a crash.  I can still force the transfer by telling fileZilla to disconnect from the server.  (Note the test files for these transfers are very small at about 20 bytes.)

I now have a work around in my code so that I can download files from my app without need for a full FTP server.

xreef, 2025-01-11:
Hi all,
I do some changes and I test ESP32 and SD in all way with Filezilla suggested configuration and seems all ok,
If you can give me a feedback.
Thanks Renzo