xreef/SimpleFTPServer #73: Transfer of larger files (>190kB) stalls

gareth3lee, 2024-12-03:
Sorry to overload the issues list.  I notice there is a problem when transferring larger files (> 190kB) as the transmission seems to stall.  (All these tests are with the linux command line ftp client (tnftp) and are therefore apparently separate from the other issue I previously posted).

I have created two test files, big.txt (112788 bytes) and bigger.txt (225576 bytes).  The first file is purely ascii textual as it was created by base64 encoding some binary data (e.g. all lines are the same length).  The second file is just a duplication of the data in the first file.

The first file transfers without error and linux ftp produces the following transcript:
```
ftp> size big.txt
big.txt	112788
ftp> get big.txt
local: big.txt remote: big.txt
227 Entering Passive Mode (10,1,1,15,195,89)
150-Connected to port 50009
150 112788 bytes to download
100% |***********************************|   110 KiB  183.10 KiB/s    00:00 ETA
226-File successfully transferred
226 580 ms, 194 kbytes/s
112788 bytes received in 00:00 (183.04 KiB/s)
```
But when I try to transfer the second bigger file I get the following error message from ftp:
```
ftp> size bigger.txt
bigger.txt	225576
ftp> get bigger.txt
local: bigger.txt remote: bigger.txt
227 Entering Passive Mode (10,1,1,15,195,89)
150-Connected to port 50009
150 225576 bytes to download
 90% |*******************************    |   200 KiB  199.94 KiB/s    00:00 ETA
ftp: Reading from network: Interrupted system call
  0% |                                   |    -1        0.00 KiB/s    --:-- ETA
426 Data connection closed. Transfer aborted
```

In this case the ESP32 server reports the following messages:
```
-S-I-Z-E- -b-i-g-g-e-r-.-t-x-t-
-
Command is: SIZE
utf8_strlen11
PATH --> /bigger.txt ...EXIST!
File to open /bigger.txt readType r
TRUE
-S-I-Z-E- -b-i-g-g-e-r-.-t-x-t-
-
Command is: SIZE
utf8_strlen11
PATH --> /bigger.txt ...EXIST!
File to open /bigger.txt readType r
TRUE
-P-A-S-V-
-
Command is: PASV
 IP: 10.1.1.15
 Soft IP: 10.1.1.15
 Connection management set to passive
 Listening at 10.1.1.15:50009
-R-E-T-R- -b-i-g-g-e-r-.-t-x-t-
-
Command is: RETR
utf8_strlen11
PATH --> /bigger.txt ...EXIST!
File to open /bigger.txt readType r
TRUE
 Sending bigger.txt size 225576
FTP ftp_transfer 2 bigger.txt 225576
NB --> 1024
FTP ftp_transfer 3 bigger.txt 1024
  Command Old: 5  Transfer Old: 0  Data Old: 1
  Command    : 5  Transfer    : 1  Data    : 1
NB --> 1024
FTP ftp_transfer 3 bigger.txt 2048
NB --> 1024
FTP ftp_transfer 3 bigger.txt 3072
NB --> 1024

<transmission of chunks uneventful>

NB --> 1024
FTP ftp_transfer 3 bigger.txt 201728
NB --> 1024
FTP ftp_transfer 3 bigger.txt 202752
NB --> 1024
FTP ftp_transfer 3 bigger.txt 203776
NB --> 1024
FTP ftp_transfer 3 bigger.txt 204800
NB --> 1024
FTP ftp_transfer 3 bigger.txt 205824
  Command Old: 5  Transfer Old: 1  Data Old: 1
  Command    : 5  Transfer    : 0  Data    : 1

<transmission halts at this point without completing the file transfer>
```

This could be a limitation in the SD/FAT library used on my ESP32, but I have certainly been able to create files larger than this on Arduinos in the past by repeatedly appending each new record to an existing file.  As it's all being processed in 1024 byte chunks I cannot see any obvious reason why transfer size would be limited in this way.

I hope the diagnostics are sufficient to give more experienced users some clue to the underlying cause.

I will investigate the source code a little further and post an update if I uncover anything that might be useful.

Thanks, in advance,

Gareth

gareth3lee, 2024-12-20:
To provide a little more context to this problem I have implemented a work around in my application code.  I have implemented a "monitor console" and have now added a unix-like cat command to allow files to be dumped from SD to the console.  It does not attempt to implement the full FTP protocol.

The cat command works in a similar way to SimpleFTPServer's RETR by transferring the file in a series of chunks using a small buffer (128 bytes in my case).  This means that it multitasks with the rest of my code (a home automation app containing an MQTT broker) and it allows file transfers of unlimited size.

It works correctly for the both big.txt and bigger.txt from my earlier post, and suggests that there are no problems with the SD card or the SD/FS libraries when accessing large files.

As I can use this work-around for the time being I will only return to investigate SimpleFTPServer when I have some spare time in the coming months.

xreef, 2025-01-11:
Hi, 
after some fixes, I try to update files bigger than 6MB, and work correctly.
Thanks for the issue, and retry when I release the new version.
Thanks Renzo

closed 2025-01-28 by xreef