xreef/SimpleFTPServer #75: Issue with long file names due to lossy type conversion in FtpServer::readChar()

mcclellansws, 2025-01-30:
https://github.com/xreef/SimpleFTPServer/blob/master/FtpServer.cpp#L1935C8-L1935C20

In the following code segment:
```

int8_t FtpServer::readChar()
{
  int8_t rc = -1;

  if( client.available())
  {
    char c = client.read();
    DEBUG_PRINT("-");
    DEBUG_PRINT( c );

    if( c == '\\' )
      c = '/';
    if( c != '\r' ){
      if( c != '\n' )
      {
        if( iCL < FTP_CMD_SIZE )
          cmdLine[ iCL ++ ] = c;
        else
          rc = -2; //  Line too long
      }
      else
      {
        cmdLine[ iCL ] = 0;
        command[ 0 ] = 0;
        parameter = NULL;
        // empty line?
        if( iCL == 0 )
          rc = 0;
        else
        {
          rc = iCL;
```

the final rc = iCL takes a unsigned 16 bit to a signed 8 bit. With long file paths the iCL can cause the rc value to roll over.

The quick fix is to change the function signature and rc variable to a signed 32 bit value

```
int32_t FtpServer::readChar()
{
  int32_t rc = -1;
```

--stevem

xreef, 2025-02-07:
Thanks, I'm going to fix. 
Bye Renzo

xreef, 2025-03-22:
Thanks, I'm going to release it soon.
Bye Renzo