xreef/SimpleFTPServer #76: SDFAT2 does not seem to compile for ESP8266 or ESP32

dreamer1234, 2025-03-16:
Hello,

When I choose STORAGE_SDFAT2, I'm getting this compilation error:

Compiling .pio\build\esp12e\lib0ed\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
src\FTPServer.cpp: In member function 'bool FtpServer::openFile(char*, int)':
src\FTPServer.cpp:2323:50: error: no match for 'operator=' (operand types are 'FsFile' and 'File32')
 2323 |   file = STORAGE_MANAGER.open( path, readTypeInt );

But If I choose STORAGE_SD, everything is fine.

I'd like to make it work with the SDFAT library since I understand is a lot faster than the SD one. The thing is that the SDFAT library has had some recent pushes with important changes.

Could you check if it still works of if it's something broken? 

Thanks so much.

xreef, 2025-03-22:
Hi,
I try this 
``` 
/*
 * Simple FTP Server Example with SD Card on ESP32
 *
 * AUTHOR: Renzo Mischianti
 * URL: https://www.mischianti.org
 *
 * DESCRIPTION:
 * This example demonstrates how to use the SimpleFTPServer library
 * with an ESP32 and an SD card module. The ESP32 connects to a WiFi network
 * and initializes an FTP server for file transfers.
 *
 * FEATURES:
 * - WiFi connection to local network
 * - SD card initialization for file storage
 * - FTP server setup for file uploads and downloads
 */

#include <WiFi.h>

#include <SimpleFTPServer.h>
#include <SdFat.h>
#include <sdios.h>

// WiFi credentials
const char* WIFI_SSID = "ssid";    // Replace with your WiFi SSID
const char* WIFI_PASSWORD = "passwd";     // Replace with your WiFi password

// SD card chip select pin
const int CHIP_SELECT_PIN = SS;               // Default SS pin for SPI

// FTP server instance
FtpServer ftpServer;

// Object for File system
SdFat sd;

void setup() {
  // Initialize Serial Monitor
  Serial.begin(9600);
  while (!Serial) {
    // Wait for serial port to connect (required for native USB ports)
  }

  // Connect to WiFi network
  Serial.println("Connecting to WiFi...");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("\nWiFi connected!");
  Serial.printf("Connected to: %s\n", WIFI_SSID);
  Serial.printf("IP Address: %s\n", WiFi.localIP().toString().c_str());

  // Wait for a short delay before initializing SD card
  delay(1000);

  // Mount the SD card memory
  Serial.print("Mount the SD card memory... ");
  if( ! sd.begin( CHIP_SELECT_PIN, SD_SCK_MHZ( 50 )))
  {
	Serial.println("unable to mount SD card");
    while( true ) ;
  }
  pinMode( CHIP_SELECT_PIN, OUTPUT );
  digitalWrite( CHIP_SELECT_PIN, HIGH );
  Serial.println("ok");

  // Show capacity and free space of SD card
  Serial.print("Capacity of card:   "); Serial.print( sd.card()->sectorCount() ); Serial.println(" kBytes");
  Serial.print("Free space on card: "); Serial.print( sd.vol()->freeClusterCount() * sd.vol()->sectorsPerCluster() >> 1 ); Serial.println(" kBytes");

  // Start FTP server with username and password
  ftpServer.begin("user", "password"); // Replace with your desired FTP credentials
  Serial.println("FTP server started!");
}

void loop() {
  // Handle FTP server operations
  ftpServer.handleFTP(); // Continuously process FTP requests
}
```

And build without problems.

Bye Renzo

dreamer1234, 2025-03-23:
Hi Renzo,

Thanks a lot for answer.
The thing is I'm using VSCode with PlatformIO. I'm thinking there's
something rare with SDFat library versions here and does not seems to
compile (at least for me). But using the SD library everything is fine.

I'll check in Arduino IDE and see if I'm having the same issue.

Thanks

dreamer1234, 2025-03-25:
Hi Renzo,

I have tried your example and it does not seem to compile with the default options.

- SDFat version and configurations:

[image]

Then, I'm compiling for an ESP32-C6 module.  Copied and pasted your example.

This is the trace:

In file included from C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original/SimpleFTPServer.h:14,
                 from C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\SimpleFTPServer_original.ino:1:
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original/FtpServer.h: In member function 'uint32_t FtpServer::capacity()':
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original/FtpServer.h:694:56: error: 'SdCard' {aka 'class SdSpiCard'} has no member named 'cardSize'
  694 |   uint32_t capacity() { return STORAGE_MANAGER.card()->cardSize() >> 1; };
      |                                                        ^~~~~~~~
In file included from C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp:50:
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original/FtpServer.h: In member function 'uint32_t FtpServer::capacity()':
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original/FtpServer.h:694:56: error: 'SdCard' {aka 'class SdSpiCard'} has no member named 'cardSize'
  694 |   uint32_t capacity() { return STORAGE_MANAGER.card()->cardSize() >> 1; };
      |                                                        ^~~~~~~~
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp: In member function 'bool FtpServer::doRetrieve()':
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp:1269:17: error: 'class SdFile' has no member named 'seek'; did you mean 'peek'?
 1269 |       if (!file.seek(restartPos)) {
      |                 ^~~~
      |                 peek
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp: In member function 'bool FtpServer::openFile(char*, int)':
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp:2370:64: error: no match for 'operator=' (operand types are 'SdFile' and 'FsFile')
 2370 |                 file = STORAGE_MANAGER.open( path, readTypeInt );
      |                                                                ^
In file included from C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original/FtpServer.h:412:
c:\Users\XXXXXXX\Documents\Arduino\libraries\SdFat\src/SdFat.h:467:7: note: candidate: 'SdFile& SdFile::operator=(SdFile&&)'
  467 | class SdFile : public PrintFile<SdBaseFile> {
      |       ^~~~~~
c:\Users\XXXXXXX\Documents\Arduino\libraries\SdFat\src/SdFat.h:467:7: note:   no known conversion for argument 1 from 'FsFile' to 'SdFile&&'
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp: In member function 'bool FtpServer::getFileModTime(uint16_t*, uint16_t*)':
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp:2518:3: error: 'dir_t' was not declared in this scope; did you mean 'div_t'?
 2518 |   dir_t d;
      |   ^~~~~
      |   div_t
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp:2520:14: error: 'class SdFile' has no member named 'dirEntry'
 2520 |   if( ! file.dirEntry( & d ))
      |              ^~~~~~~~
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp:2520:26: error: 'd' was not declared in this scope; did you mean 'sd'?
 2520 |   if( ! file.dirEntry( & d ))
      |                          ^
      |                          sd
C:\Users\XXXXXXX\Documents\PlatformIO\Projects\SimpleFTPServer_original\FtpServer.cpp:2522:13: error: 'd' was not declared in this scope; did you mean 'sd'?
 2522 |   * pdate = d.lastWriteDate;
      |             ^
      |             sd

exit status 1

Compilation error: 'SdCard' {aka 'class SdSpiCard'} has no member named 'cardSize'

dreamer1234, 2025-03-25:
Just to update. It does compile on Arduino IDE with the right libraries.

Thanks a lot Renzo!!

xreef, 2025-03-27:
Good! Thanks for your feedback.

dreamer1234, 2025-03-27:
Hi Renzo,

What I don't understand is why when you are using sdfat does not compile on VScode even with the lib_deps line like described here:
https://registry.platformio.org/libraries/greiman/SdFat/installation

I understand this is something related to VSCode and PlatformIO but is something really strange.

Thanks